unitreerobotics/unilidar_sdk #6: Documentation clarification

DevonMorris, 2023-09-02:
I just ordered one of your lidar and am preparing to set up a ROS environment to use the lidar. Upon reviewing your SDK, I had a few questions.

* What is `cloud_scan_num` and why is it default set to 18?
* How do I interpret the different modes in `LEDDisplayMode` and do these effect the scan pattern of the lidar?
* How do I interpret the `time` field of `PointUnitree`? Why is this a floating point number? Does this time field have units? Similarly for `stamp` in `PointCloudUnitree`

Thanks for your support. I'm excited to get the hardware in hand and use your lidar!

bluecamel, 2023-09-14:
I've also been getting familiar with the SDK recently and I can answer some of these questions.

>     * What is `cloud_scan_num` and why is it default set to 18?

I'd also like some clarification on this.  I'd guess that they're doing something similar to what spinning, multi-channel lidar manufacturers do, where each channel has a ring ID.  Ouster and Velodyne lidars that rotate 360 degrees and have some number of vertical channels/beams would have a corresponding number of rings in the output cloud.  For example, a 64 channel lidar would produce 64 rings, a 128 channel would produce 128 rings, and so on.

That said, I'm not sure how that translates to this lidar since 1) there aren't truly distinct channels/beams/rings, and 2) I think there would be more than 18.  If I had to guess, I'd say that they probably split the cloud up into 18 regions, but I think that would mean every 5 degrees of vertical FOV.  Or perhaps it's every 10 degrees of the full 180, but that seems even more strange.

I just got the ROS node working with my unit last night.  Sometime in the next week, I can look closer at the data and see if I can tease out how that works here.

>     * How do I interpret the different modes in `LEDDisplayMode` and do these effect the scan pattern of the lidar?

The English documentation is surprisingly hard to find on their site, but there's a bit more information in both the user manual (page 11) and unilidar user manual (page 5).

FWIW, I've found these documents in English:

- [L1 User Manual](https://oss-global-cdn.unitree.com/static/f5ca3e9038d44ca29f711f90005a2185.pdf)
- [L1 Quick Start Guide](https://unitree-website.oss-cn-hangzhou.aliyuncs.com/Lidar/L1%20Quike%20Start%20Guide_v1.0.pdf)
- [Unilidar User Manual](https://unitree-website.oss-cn-hangzhou.aliyuncs.com/Lidar/Unilidar%20User%20Manual_v1.0.pdf)
- [Unilidar SDK User Manual](https://unitree-website.oss-cn-hangzhou.aliyuncs.com/Lidar/Unilidar%20SDK%20User%20Manual_v1.0.pdf)

>     * How do I interpret the `time` field of `PointUnitree`? Why is this a floating point number? Does this time field have units? Similarly for `stamp` in `PointCloudUnitree`

I'm certain of one thing here, which is that the time field is an offset for each point.  The stamp of the overall point cloud will be the beginning time and the time for each point is an offset from the point cloud stamp.  Now, I'm not yet sure if this is in milliseconds, microseconds, or nanoseconds.  I would suspect that it's not the latter since it's a float.

lingbomeng, 2023-09-20:
Hi guys,

The `cloud_scan_num` means how many scan lines are cached as a frame of pointcloud. For example, if you set it to 18, then a frame of pointcloud will contain 18 laser scans. So the higher `cloud_scan_num` is, the lower the cloud frequency will be, which is because the total scans per second is fixed.

Secondly, the `LEDDisplayMode` is used only for controlling the LED lights' pattern, which is only for visual effect. So this will not effect the scan pattern of the lidar.

Lastly, the `time` field of `PointUnitree` stands for the relative time of this point from the cloud stamp and its unit is second.

gh0st-gh, 2024-01-25:
What are the actual numbers though? "The higher it is, the slower" doesn't say much. The code is all closed source apparently.  I don't think it's asking too much for a $400+ product to give configuration parameters and a datasheet, instead of requiring everyone to do trial and error and graph things just to figure out how they work.

Also, exactly how doe these parameters work?
      float rotate_yaw_bias = 0,
      float range_scale = 0.001, 
      float range_bias = 0,
      float range_max = 50,
      float range_min = 0
  
I assume the range is in meters, but exactly how do the rest work?  How does the yaw bias work?  Can it be used to counteract rotational motion of the base, and by what factor per speed unit?
What are the units of range scale?
What's range bias and what are its UNITS?

quillaja, 2024-10-07:
According to https://oss-global-cdn.unitree.com/Lidar/Unilidar%20SDK%20User%20Manual_v1.0.pdf (in the python section) the parameters have these meanings:
```
port：The serial port number for connecting the LiDAR to the computer is /dev/ttyUSB0 by default.
rotate_yaw_bias：The fixed offset angle in degrees that may exist during the installation of the LiDAR. This parameter is used to calibrate the orientation of the point cloud data, and its default value is 0.
range_scale：The range scaling factor of the LiDAR, which is used to calibrate the ranging accuracy of the LiDAR. Its default value is 0.001.
range_bias：The range offset of the LiDAR, which is used to calibrate the ranging accuracy of the LiDAR. Its default value is 0.
range_max：The maximum range of the LiDAR, measured in meters, with a default value of 50.
range_min：The minimum range of the LiDAR, measured in meters, with a default value of 0.
cloud_scan_num：The number of laser ring IDs for the point cloud data, with a default value of 18.
```

I'm evaluating if I want to buy one of these units, and was examining their example rosbag file. It appears that each rosbag "message" is 18 "rings", whatever those are (1 ring=1 rotation around Z?), and corresponds to the `cloud_scan_num`. This matches what @lingbomeng said. If I take the number of points in each rosbag message (usually round 2032) and divide by the time of the last point (around 0.01s), you'll get a points/sec frequency about the stated 21.6K points/s. Eg: 2046/0.097276 = 21032.9.

After parsing all the points in their example rosbag file, I converted them to XYZ and plotted in Meshlab as one large pointcloud. The sample appears to be an interior space (office?), and the lidar appears to be mounted with Z horizontal ("front") and Z down at the floor. The main thing that is a little concerning is that there are visible spherical artifacts above what appears to be the ceiling and below what appears to be the floor. I will have to examine the data more to see if that's a problem or not.