unitreerobotics/unilidar_sdk #7: SLAM - Lidar stability while spinning

Seroka88, 2023-11-09:
Hey,

i ported some great SLAM frameworks like FAST-LIO2 and PV-LIO to run with your lidar. Your bag file is looking very good.

My question is, in your example you are walking around an office. Did you mount the lidar to any heavy device or structure?

My unilidar is spinning like crazy while just holding it in my hand and walking around. The IMU is getting some crazy measurements while walking around.

Do you have any tips / advice for stability?

Kind regards
Seroka

Seroka88, 2023-11-10:
[image]

jasonwjw, 2023-12-14:
Hi, Seroka

I am new to ROS and slam, I tried to port the FAST-LIO2 to run with this UniTree 4D L1 lidar on NVIDIA Jetson Orin Nano.  I followed their official procedure and tried to compile the FAST-LIO2.  But they need the livox_ros_driver  and I doubted whether it was necessary for Unitree 4D L1 lidar. Could you please give me some guidance? 

Many thanks
Jason